mikaelnousiainen/RS41ng #1: region `rom' overflowed by 4 bytes

fkemps, 2021-09-07:
Hello,

While building RS41ng I run in `region from" overflowed by 4 bytes` error message while linking.
Platform is Ubuntu 20.04.2 LTS.

Any hint how to solve?

```
➜  build git:(master) cmake ..
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/fkemps/RS41ng/build
➜  build git:(master) make
Scanning dependencies of target RS41ng.elf
[  1%] Building C object src/CMakeFiles/RS41ng.elf.dir/bmp280_handler.c.o
[  2%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/aprs/aprs.c.o
[  4%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/ax25/ax25.c.o
[  5%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/bell/bell.c.o
[  6%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/horus/horus_l2.c.o
[  8%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/horus/horus_packet_v1.c.o
[  9%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/jtencode/lib/crc14.c.o
[ 10%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/mfsk/mfsk.c.o
[ 12%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/morse/morse.c.o
[ 13%] Building C object src/CMakeFiles/RS41ng.elf.dir/config.c.o
[ 15%] Building C object src/CMakeFiles/RS41ng.elf.dir/drivers/bmp280/bmp280.c.o
[ 16%] Building C object src/CMakeFiles/RS41ng.elf.dir/drivers/si4032/si4032.c.o
[ 17%] Building C object src/CMakeFiles/RS41ng.elf.dir/drivers/ubxg6010/ubxg6010.c.o
[ 19%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/cmsis_boot/startup/startup_stm32f10x_md_vl.c.o
[ 20%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/cmsis_boot/system_stm32f10x.c.o
[ 21%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/datatimer.c.o
[ 23%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/delay.c.o
[ 24%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/i2c.c.o
[ 26%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/pwm.c.o
[ 27%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/spi.c.o
[ 28%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/misc.c.o
[ 30%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_adc.c.o
[ 31%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_dma.c.o
[ 32%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_flash.c.o
[ 34%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_gpio.c.o
[ 35%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_i2c.c.o
[ 36%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_pwr.c.o
[ 38%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_rcc.c.o
[ 39%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_spi.c.o
[ 41%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_tim.c.o
[ 42%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_usart.c.o
[ 43%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/system.c.o
[ 45%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/usart_ext.c.o
[ 46%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/usart_gps.c.o
[ 47%] Building C object src/CMakeFiles/RS41ng.elf.dir/locator.c.o
[ 49%] Building C object src/CMakeFiles/RS41ng.elf.dir/log.c.o
[ 50%] Building C object src/CMakeFiles/RS41ng.elf.dir/main.c.o
[ 52%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio.c.o
[ 53%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_aprs.c.o
[ 54%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_cw.c.o
[ 56%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_fsq.c.o
[ 57%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_horus_v1.c.o
[ 58%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_jtencode.c.o
[ 60%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_wspr.c.o
[ 61%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_si4032.c.o
[ 63%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_si5351.c.o
[ 64%] Building C object src/CMakeFiles/RS41ng.elf.dir/syscalls/semihosting.c.o
[ 65%] Building C object src/CMakeFiles/RS41ng.elf.dir/syscalls/syscalls.c.o
[ 67%] Building C object src/CMakeFiles/RS41ng.elf.dir/telemetry.c.o
[ 68%] Building C object src/CMakeFiles/RS41ng.elf.dir/template.c.o
[ 69%] Building C object src/CMakeFiles/RS41ng.elf.dir/utils.c.o
[ 71%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/codecs/jtencode/jtencode.cpp.o
[ 72%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/codecs/jtencode/lib/JTEncode.cpp.o
[ 73%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/codecs/jtencode/lib/encode_rs_int.cpp.o
[ 75%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/codecs/jtencode/lib/init_rs_int.cpp.o
/home/fkemps/RS41ng/src/codecs/jtencode/lib/init_rs_int.cpp: In member function 'void* JTEncode::init_rs_int(int, int, int, int, int, int)':
/home/fkemps/RS41ng/src/codecs/jtencode/lib/init_rs_int.cpp:33:29: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
   33 |   if(symsize < 0 || symsize > 8*sizeof(data_t)){
      |                     ~~~~~~~~^~~~~~~~~~~~~~~~~~
[ 76%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/drivers/si5351/si5351.cpp.o
[ 78%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/si5351_handler.cpp.o
[ 79%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/si5351_test.cpp.o
[ 80%] Linking CXX executable RS41ng.elf
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: RS41ng.elf section `.ARM.exidx' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 4 bytes
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/RS41ng.elf.dir/build.make:939: src/RS41ng.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:143: src/CMakeFiles/RS41ng.elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

kareiva, 2021-10-13:
Also similar case on Ubuntu 20.04 / WSL here:

```
Scanning dependencies of target RS41ng.elf
[  1%] Building C object src/CMakeFiles/RS41ng.elf.dir/bmp280_handler.c.o
[  2%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/aprs/aprs.c.o
[  3%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/horus/horus_packet_v1.c.o
[  5%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/horus/horus_packet_v2.c.o
[  6%] Building C object src/CMakeFiles/RS41ng.elf.dir/config.c.o
[  7%] Building C object src/CMakeFiles/RS41ng.elf.dir/drivers/ubxg6010/ubxg6010.c.o
[  8%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/pwm.c.o
[ 10%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/system.c.o
[ 11%] Building C object src/CMakeFiles/RS41ng.elf.dir/log.c.o
[ 12%] Building C object src/CMakeFiles/RS41ng.elf.dir/main.c.o
[ 13%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio.c.o
[ 15%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_aprs.c.o
[ 16%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_cw.c.o
[ 17%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_fsq.c.o
[ 18%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_horus_v1.c.o
[ 20%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_horus_v2.c.o
[ 21%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_jtencode.c.o
[ 22%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_payload_wspr.c.o
[ 24%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_si4032.c.o
[ 25%] Building C object src/CMakeFiles/RS41ng.elf.dir/radio_si5351.c.o
[ 26%] Building C object src/CMakeFiles/RS41ng.elf.dir/syscalls/syscalls.c.o
[ 27%] Building C object src/CMakeFiles/RS41ng.elf.dir/telemetry.c.o
[ 29%] Building C object src/CMakeFiles/RS41ng.elf.dir/template.c.o
[ 30%] Linking CXX executable RS41ng.elf
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: RS41ng.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 856 bytes
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/RS41ng.elf.dir/build.make:999: src/RS41ng.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:143: src/CMakeFiles/RS41ng.elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

mikaelnousiainen, 2021-11-11:
The options for enabling/disabling specific modes now also remove compiled code accordingly, so that the firmware will become smaller when disabling modes.

The default config without Si5351 should always produce a binary that fits the sonde flash memory.

closed 2021-11-11 by mikaelnousiainen