mikaelnousiainen/RS41ng #4: BMP 280

aledorrego, 2021-11-11:
hello.  I was reading that for the BMP280 it uses pull up resistors.  but where are they connected?  I haven't found a diagram yet.  I'm about to upload the latest update.  thank you, very good job.

mikaelnousiainen, 2021-11-11:
@aledorrego The README file indicates the following pinout:

```
______________________|           |______________________
|                                                       |
|   1           2           3           4           5   |
|                                                       |
|   6           7           8           9          10   |
|_______________________________________________________|
```

(View from the bottom of the sonde, pay attention to the gap in the connector)

Connect your I²C device (e.g. BMP280) to the following pins:

3 - MCU switched 3.3V out to external device
4 - I2C2_SCL -- This is the external I²C port clock pin for Si5351 and sensors
5 - GND
10 - I2C2_SDA -- This is the external I²C port data pin for Si5351 and sensors

mikaelnousiainen, 2021-11-11:
@aledorrego Also, the BMP280 readings are only available if you specifically add them to one of the following: APRS comment text, CW text or FSQ message. Horus V2 includes them in custom data. Other modes do not have this info directly.

See the config.c file for defining the text messages for each mode.

closed 2021-11-11 by aledorrego