mikaelnousiainen/RS41ng #5: APRS

aledorrego, 2021-11-14:
Hi. What part of the program should be modified so that in APRS.FI when I click on the probe, show the temperature, humidity and pressure on the balloon that appears on the station? He currently does not show me anything. Figure as LW7EEA-13

mikaelnousiainen, 2021-11-14:
@aledorrego The RS41ng firmware currently supports only location APRS messages, which contain the temperature, humidity and pressure information in the APRS comment field as plain text (not in any machine-readable telemetry format). Weather station telemetry, which aprs.fi expects when using the weather station APRS symbol, is _not_ yet supported.

It seems the comment field is not displayed by aprs.fi if you set the APRS symbol to be a weather station `_`. The comment field you've specified is visible in the raw messages here: https://aprs.fi/?c=raw&call=LW7EEA-13

If you set the symbol to something else (maybe like a home symbol?), you should be able to view the APRS comment in aprs.fi.

That said, I could add a feature in the firmware to publish weather station telemetry in the standard format, it should not be too difficult.

Additionally, it seems that the atmospheric pressure is displayed incorrectly in the APRS comment, so there's a small bug in the code.

aledorrego, 2021-11-14:
good perfect.  I'm going to try another symbol in aprs.  As for the pressure value, I think my sensor is damaging due to excess temperature when soldering it.  in the week a BME280 arrives and I will try.

mikaelnousiainen, 2021-11-16:
@aledorrego There's now an option to transmit APRS weather reports, see `config.h`. The bug with atmospheric pressure readings is also fixed.

closed 2021-11-16 by mikaelnousiainen

aledorrego, 2021-11-16:
hello.  there load the update.  It works fine for me in weather station mode, it shows all the data.  but it shows them at zero.  the probe does not recognize the sensor.  I go back to previous builds and the sensor works fine for me.  change something in this update?  Thanks