mikaelnousiainen/RS41ng #6: Sensor data problem

aledorrego, 2021-11-17:
hello again, hope not to disturb. I loaded the last update and I found that it does not recognize the sensor, the probe takes 15 seconds to start, it takes 35 seconds, and it does not throw the sensor data. Could something have happened in the code? I was able to visualize with the weather station logo but with zero values. Sorry for so many questions, and there is very little to work. Thank you

mikaelnousiainen, 2021-11-18:
@aledorrego Can you post your _new_ config.h file contents here so we can check what is enabled/disabled? I can try to test the same config locally.

aledorrego, 2021-11-18:
#ifndef __CONFIG_H
#define __CONFIG_H

// Enable semihosting to receive debug logs during development
// NOTE: Semihosting has to be disabled when the RS41 radiosonde is not connected to the STM32 programmer dongle, otherwise the firmware will not run.
//#define SEMIHOSTING_ENABLE
//#define LOGGING_ENABLE

/**
 * Global configuration
 */

// Set the tracker amateur radio call sign here
#define CALLSIGN "LW7EEA"

// Disabling LEDs will save power
// Red LED: Lit during initialization and transmit.
// Green LED: Blinking fast when there is no GPS fix. Blinking slowly when the GPS has a fix.
#define LEDS_ENABLE true

// Allow powering off the sonde by pressing the button for over a second (when the sonde is not transmitting)
#define ALLOW_POWER_OFF true

// Enable use of an externally connected I²C BMP280 atmospheric sensor
#define SENSOR_BMP280_ENABLE true

// Enable use of an externally connected I²C Si5351 clock generator chip for HF radio transmissions
#define RADIO_SI5351_ENABLE false

// Number of character pairs to include in locator
#define LOCATOR_PAIR_COUNT_FULL 6 // max. 6 (12 characters WWL)

// Delay after transmission for modes that do not use time synchronization. Zero delay allows continuous transmit mode for Horus V1 and V2.
#define RADIO_POST_TRANSMIT_DELAY_MS 60000

// Threshold for time-synchronized modes regarding how far from scheduled transmission time the transmission is still allowed
#define RADIO_TIME_SYNC_THRESHOLD_MS 2000

// Enable NMEA output from GPS via external serial port. This disables use of I²C bus (Si5351 and sensors) because the pins are shared.
#define GPS_NMEA_OUTPUT_VIA_SERIAL_PORT_ENABLE false

#if (GPS_NMEA_OUTPUT_VIA_SERIAL_PORT_ENABLE) && ((RADIO_SI5351_ENABLE) || (SENSOR_BMP280_ENABLE))
#error GPS NMEA output via serial port cannot be enabled simultaneously with the I2C bus.
#endif

/**
 * Built-in Si4032 radio chip transmission configuration
 */

// Si4032 transmit power: 0..7
// 0 = -1dBm, 1 = 2dBm, 2 = 5dBm, 3 = 8dBm, 4 = 11dBm, 5 = 14dBm, 6 = 17dBm, 7 = 20dBm
#define RADIO_SI4032_TX_POWER 6

// Which modes to transmit using the built-in Si4032 transmitter chip
#define RADIO_SI4032_TX_CW true
#define RADIO_SI4032_TX_APRS true
#define RADIO_SI4032_TX_HORUS_V1 false
#define RADIO_SI4032_TX_HORUS_V2 false

// Continuous transmit mode can be enabled for *either* Horus V1 or V2, but not both. This disables all other transmission modes.
// The continuous mode transmits Horus 4FSK preamble between transmissions
// to allow Horus receivers to keep frequency synchronization at all times, which improves reception.
#define RADIO_SI4032_TX_HORUS_V1_CONTINUOUS false
#define RADIO_SI4032_TX_HORUS_V2_CONTINUOUS false

// Transmit frequencies for the Si4032 transmitter modes
#define RADIO_SI4032_TX_FREQUENCY_CW        430050600
#define RADIO_SI4032_TX_FREQUENCY_APRS_1200 430930000
// Use a frequency offset to place FSK tones slightly above the defined frequency for SSB reception
#define RADIO_SI4032_TX_FREQUENCY_HORUS_V1  432501000
[... 141 lines cut ...]

#define FSQ_TIME_SYNC_SECONDS 0
#define FSQ_TIME_SYNC_OFFSET_SECONDS 0

/**
 * FT/JT mode settings
 */

// Schedule transmission every 15 seconds, counting from beginning of an hour (based on GPS time). Set to zero to disable time sync.
#define FT8_TIME_SYNC_SECONDS 15
// Delay transmission for 1 second after the scheduled time.
#define FT8_TIME_SYNC_OFFSET_SECONDS 1

#define JT9_TIME_SYNC_SECONDS 60
#define JT9_TIME_SYNC_OFFSET_SECONDS 1

#define JT4_TIME_SYNC_SECONDS 60
#define JT4_TIME_SYNC_OFFSET_SECONDS 1

#define JT65_TIME_SYNC_SECONDS 60
#define JT65_TIME_SYNC_OFFSET_SECONDS 1

#include "config_internal.h"

#endif

aledorrego, 2021-11-18:
//does not send sensor data neither in cw nor aprs. it pretends not to recognize the sensor. will the change have something to do with using nmea gps output?

#ifndef __CONFIG_H
#define __CONFIG_H

// Enable semihosting to receive debug logs during development
// NOTE: Semihosting has to be disabled when the RS41 radiosonde is not connected to the STM32 programmer dongle, otherwise the firmware will not run.
//#define SEMIHOSTING_ENABLE
//#define LOGGING_ENABLE

/**
 * Global configuration
 */

// Set the tracker amateur radio call sign here
#define CALLSIGN "LW7EEA"

// Disabling LEDs will save power
// Red LED: Lit during initialization and transmit.
// Green LED: Blinking fast when there is no GPS fix. Blinking slowly when the GPS has a fix.
#define LEDS_ENABLE true

// Allow powering off the sonde by pressing the button for over a second (when the sonde is not transmitting)
#define ALLOW_POWER_OFF true

// Enable use of an externally connected I²C BMP280 atmospheric sensor
#define SENSOR_BMP280_ENABLE true

// Enable use of an externally connected I²C Si5351 clock generator chip for HF radio transmissions
#define RADIO_SI5351_ENABLE false

// Number of character pairs to include in locator
#define LOCATOR_PAIR_COUNT_FULL 6 // max. 6 (12 characters WWL)

// Delay after transmission for modes that do not use time synchronization. Zero delay allows continuous transmit mode for Horus V1 and V2.
#define RADIO_POST_TRANSMIT_DELAY_MS 60000

// Threshold for time-synchronized modes regarding how far from scheduled transmission time the transmission is still allowed
#define RADIO_TIME_SYNC_THRESHOLD_MS 2000

// Enable NMEA output from GPS via external serial port. This disables use of I²C bus (Si5351 and sensors) because the pins are shared.
#define GPS_NMEA_OUTPUT_VIA_SERIAL_PORT_ENABLE false

#if (GPS_NMEA_OUTPUT_VIA_SERIAL_PORT_ENABLE) && ((RADIO_SI5351_ENABLE) || (SENSOR_BMP280_ENABLE))
#error GPS NMEA output via serial port cannot be enabled simultaneously with the I2C bus.
#endif

/**
 * Built-in Si4032 radio chip transmission configuration
 */

// Si4032 transmit power: 0..7
// 0 = -1dBm, 1 = 2dBm, 2 = 5dBm, 3 = 8dBm, 4 = 11dBm, 5 = 14dBm, 6 = 17dBm, 7 = 20dBm
#define RADIO_SI4032_TX_POWER 6

// Which modes to transmit using the built-in Si4032 transmitter chip
#define RADIO_SI4032_TX_CW true
#define RADIO_SI4032_TX_APRS true
#define RADIO_SI4032_TX_HORUS_V1 false
#define RADIO_SI4032_TX_HORUS_V2 false

// Continuous transmit mode can be enabled for *either* Horus V1 or V2, but not both. This disables all other transmission modes.
// The continuous mode transmits Horus 4FSK preamble between transmissions
// to allow Horus receivers to keep frequency synchronization at all times, which improves reception.
#define RADIO_SI4032_TX_HORUS_V1_CONTINUOUS false
#define RADIO_SI4032_TX_HORUS_V2_CONTINUOUS false

// Transmit frequencies for the Si4032 transmitter modes
#define RADIO_SI4032_TX_FREQUENCY_CW        430050600
#define RADIO_SI4032_TX_FREQUENCY_APRS_1200 430930000
[... 143 lines cut ...]

#define FSQ_TIME_SYNC_SECONDS 0
#define FSQ_TIME_SYNC_OFFSET_SECONDS 0

/**
 * FT/JT mode settings
 */

// Schedule transmission every 15 seconds, counting from beginning of an hour (based on GPS time). Set to zero to disable time sync.
#define FT8_TIME_SYNC_SECONDS 15
// Delay transmission for 1 second after the scheduled time.
#define FT8_TIME_SYNC_OFFSET_SECONDS 1

#define JT9_TIME_SYNC_SECONDS 60
#define JT9_TIME_SYNC_OFFSET_SECONDS 1

#define JT4_TIME_SYNC_SECONDS 60
#define JT4_TIME_SYNC_OFFSET_SECONDS 1

#define JT65_TIME_SYNC_SECONDS 60
#define JT65_TIME_SYNC_OFFSET_SECONDS 1

#include "config_internal.h"

#endif

mikaelnousiainen, 2021-11-18:
Config looks good. I can do some more testing locally with a BMP280. The code did work for me before I committed it to GitHub :crossed_fingers:

aledorrego, 2021-11-21:
hello.  I've been testing builds prior to the weather station update and the sensor is working fine.  in fact at this moment I appear in aprs.fi . but with wx firmware it no longer read sensor data.  greetings LW7EEA

aledorrego, 2021-11-26:
Hi, friend.  could you find if there was any problem in the code?  or will I have a local problem with my program?  Cheers!!

mikaelnousiainen, 2021-11-26:
@aledorrego I tested the current code with your config file and it seems to work for me just fine:

```
LW7EEA-13 audio level = 16(1/1)   [NONE]   ____|____
[0.4] LW7EEA-13>APZ41N,WIDE1-1,WIDE2-1:/000408z0000.00S/00000.00W_.../...g...t073h28b09766 JJ00AA00AA00 - RS41 sonda 16/11/21
Weather Report, WEATHER Station (blue), Experimental
N 00 -0.0000, E 000 -0.0000
, temperature 73, humidity 28, barometer 28.84, " JJ00AA00AA00 - RS41 sonda 16/11/21"
```

Are you sure that you don't have any other changes in your code, except for the config.h file?

aledorrego, 2021-11-26:
around here we continue testing. The last thing we did was just modify the config.h and compile in Fedora, before we used Ubuntu. The following message appears. Still not getting sensor data

[root@fedora build]# cmake ..
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 11.2.1
-- The CXX compiler identification is GNU 11.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /root/RS41ng/build
[root@fedora build]# make
[  1%] Building C object src/CMakeFiles/RS41ng.elf.dir/bmp280_handler.c.o
[  2%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/aprs/aprs.c.o
[  3%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/aprs/aprs_position.c.o
[  4%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/aprs/aprs_weather.c.o
[  5%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/ax25/ax25.c.o
[  7%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/bell/bell.c.o
[  8%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/horus/horus_common.c.o
[  9%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/horus/horus_l2.c.o
[ 10%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/horus/horus_packet_v1.c.o
[ 11%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/horus/horus_packet_v2.c.o
[ 13%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/jtencode/lib/crc14.c.o
[ 14%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/mfsk/mfsk.c.o
[ 15%] Building C object src/CMakeFiles/RS41ng.elf.dir/codecs/morse/morse.c.o
[ 16%] Building C object src/CMakeFiles/RS41ng.elf.dir/config.c.o
[ 17%] Building C object src/CMakeFiles/RS41ng.elf.dir/drivers/bmp280/bmp280.c.o
[ 19%] Building C object src/CMakeFiles/RS41ng.elf.dir/drivers/si4032/si4032.c.o
[ 20%] Building C object src/CMakeFiles/RS41ng.elf.dir/drivers/ubxg6010/ubxg6010.c.o
[ 21%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/cmsis_boot/startup/startup_stm32f10x_md_vl.c.o
[ 22%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/cmsis_boot/system_stm32f10x.c.o
[ 23%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/datatimer.c.o
[ 25%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/delay.c.o
[ 26%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/i2c.c.o
[ 27%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/pwm.c.o
[ 28%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/spi.c.o
[ 29%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/misc.c.o
[ 30%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_adc.c.o
[ 32%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_dma.c.o
[ 33%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_flash.c.o
[ 34%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_gpio.c.o
[ 35%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_i2c.c.o
[ 36%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_pwr.c.o
[ 38%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_rcc.c.o
[ 39%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_spi.c.o
[ 40%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_tim.c.o
[ 41%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/stm_lib/src/stm32f10x_usart.c.o
[ 42%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/system.c.o
[ 44%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/usart_ext.c.o
[ 45%] Building C object src/CMakeFiles/RS41ng.elf.dir/hal/usart_gps.c.o
[... 36 lines cut ...]
[ 78%] Built target RS41ng.elf
[ 79%] Building C object tests/CMakeFiles/Project.dir/bell_test.c.o
[ 80%] Building C object tests/CMakeFiles/Project.dir/morse_test.c.o
[ 82%] Building C object tests/CMakeFiles/Project.dir/template_test.c.o
[ 83%] Building C object tests/CMakeFiles/Project.dir/__/src/codecs/aprs/aprs.c.o
[ 84%] Building C object tests/CMakeFiles/Project.dir/__/src/codecs/aprs/aprs_position.c.o
[ 85%] Building C object tests/CMakeFiles/Project.dir/__/src/codecs/aprs/aprs_weather.c.o
[ 86%] Building C object tests/CMakeFiles/Project.dir/__/src/codecs/ax25/ax25.c.o
[ 88%] Building C object tests/CMakeFiles/Project.dir/__/src/codecs/bell/bell.c.o
[ 89%] Building C object tests/CMakeFiles/Project.dir/__/src/codecs/horus/horus_common.c.o
[ 90%] Building C object tests/CMakeFiles/Project.dir/__/src/codecs/horus/horus_l2.c.o
[ 91%] Building C object tests/CMakeFiles/Project.dir/__/src/codecs/horus/horus_packet_v1.c.o
[ 92%] Building C object tests/CMakeFiles/Project.dir/__/src/codecs/horus/horus_packet_v2.c.o
[... cut ...]
[ 97%] Building C object tests/CMakeFiles/Project.dir/__/src/template.c.o
/root/RS41ng/src/template.c: In function ‘template_replace’:
/root/RS41ng/src/template.c:16:5: warning: implicit declaration of function ‘strlcpy’; did you mean ‘strncpy’? [-Wimplicit-function-declaration]
   16 |     strlcpy(replacement, CALLSIGN, sizeof(replacement));
      |     ^~~
      |     strncpy
[ 98%] Building C object tests/CMakeFiles/Project.dir/__/src/utils.c.o
[100%] Linking C executable Project
[100%] Built target Project

mikaelnousiainen, 2021-11-27:
Sorry to hear about the sensor trouble. I'm not able to replicate the issue here -- at least for now.

The latest change with the APRS weather station code does not include any changes to reading the sensor or I²C in general...

Have you verified that you have got the pull-up resistors required by the I²C bus applied to bot SCL (clock) and SDA (data) lines? Many BMP280 break-out boards have them, but it's worth checking.

The warning you get during compile is expected and it comes from some test code I have in the repository, not from the actual firmware compilation.

mikaelnousiainen, 2021-11-27:
You can try to enable debug logs by following the "Developing / debugging the firmware" instructions in the README file.

That will allow you to get detailed text output about what the firmware does and sends out -- you will need to run the firmware with the (ST-LINK) programmer connected to your computer.

Please let the sonde transmit a couple of APRS packets and paste the full text output here.

aledorrego, 2021-11-28:
Well, I'm going to leave it as I have it working now, with the tower symbol it sends the data well. I tried to install a virtual machine to access with fedora, but my cpu does not support virtual machine. I appreciate all your effort to help me this time, but I have certain technological limitations here. A hug

leon5265, 2022-02-05:
Same problem here..

mikaelnousiainen, 2022-11-21:
Closing issue with no activity. We have several people that have confirmed BMP280 and BME280 working with the firmware.

Some sensors seem to be picky about the I²C bus speed, so you might want to reduce I²C bus speed to 10000 (= 10 kHz) using the `I2C_BUS_CLOCK_SPEED` setting in `config.h`:

https://github.com/mikaelnousiainen/RS41ng/blob/873e281372f5cd207bac661f56d33f5fcbd99564/src/config.h#L30

closed 2022-11-21 by mikaelnousiainen