mikaelnousiainen/RS41ng #7: Compile error

Reid-n0rc, 2022-01-05:
I get an error when trying to compile. I have installed libbsd-dev and that helped with a previous error that `<bsd/string.h>` could not be found. It was my understanding that `<bsd/string.h>` contained the function `strlcpy` but it isn't working in `template.c`

```[ 96%] Building C object tests/CMakeFiles/Project.dir/__/src/codecs/morse/morse.c.o
[ 97%] Building C object tests/CMakeFiles/Project.dir/__/src/template.c.o
/home/reid/RS41ng/src/template.c: In function ‘template_replace’:
/home/reid/RS41ng/src/template.c:16:5: warning: implicit declaration of function ‘strlcpy’; did you mean ‘strncpy’? [-Wimplicit-function-declaration]
     strlcpy(replacement, CALLSIGN, sizeof(replacement));
     ^~~~~~~
     strncpy
[ 98%] Building C object tests/CMakeFiles/Project.dir/__/src/utils.c.o
[100%] Linking C executable Project
CMakeFiles/Project.dir/template_test.c.o: In function `main':
template_test.c:(.text+0x74): undefined reference to `strlcpy'
CMakeFiles/Project.dir/__/src/template.c.o: In function `template_replace':
template.c:(.text+0x61): undefined reference to `strlcpy'
template.c:(.text+0x7d): undefined reference to `strlcpy'
template.c:(.text+0xc0): undefined reference to `strlcpy'
template.c:(.text+0xdc): undefined reference to `strlcpy'
CMakeFiles/Project.dir/__/src/template.c.o:template.c:(.text+0x11f): more undefined references to `strlcpy' follow
collect2: error: ld returned 1 exit status
tests/CMakeFiles/Project.dir/build.make:510: recipe for target 'tests/Project' failed
make[2]: *** [tests/Project] Error 1
CMakeFiles/Makefile2:173: recipe for target 'tests/CMakeFiles/Project.dir/all' failed
make[1]: *** [tests/CMakeFiles/Project.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2```

kmihaylov, 2022-01-22:
Hello. I also experience the similar problem. I installed ```libbsd-dev``` on my Ununtu 18.04 with no luck.
I tried the following suggestion w/o luck:

> You can compile unmodified BSD source code by adding the following to your CFLAGS: -DLIBBSD_OVERLAY -I/usr/include/bsd, and linking with -lbsd.

https://stackoverflow.com/questions/61998682/making-strlcpy-available-in-linux

I'm not quite sure how ```strlcpy``` will reference to ```libbsd``` while the remaining parts will reference to ```newlib```. 

Also just for reference:
https://packages.ubuntu.com/bionic/amd64/libbsd-dev/filelist

kbarilar, 2022-02-16:
Hi, 

In file that is giving you warning regarding strlcpy, you should add #include <string.h>
at the beginning of that file. This will  enable usage of strlcpy. At least it worked for me. 
Don't forget to run cmake .. after this change and then make.

mikaelnousiainen, 2022-02-18:
@kmihaylov @kbarilar That sounds weird. I added instructions to install `libbsd-dev`/`libbsd-devel` to fix this. It works on my Fedora installation.

Also, `strlcpy()` does exist in the embedded libc library and where it's failing is only for the test code (`template_test.c`), which is of course not required for compiling the firmware binary. But if you come up with a solution, please share your changes to the code here or in a PR!

mikaelnousiainen, 2022-02-20:
So just to answer @kmihaylov too: The sonde firmware links with newlib indeed and it has got strlcpy() and it works correctly. The problem you see is with the **test** code that exercises the templating functionality. That will of course not use newlib and will need libbsd.

mikaelnousiainen, 2022-02-20:
Also, even if compiling the **test** code fails, you **already** have the firmware built successfully and you can flash the binary. Tests are built **after** the firmware binary, so this issue doesn't prevent from flashing a sonde.

kmihaylov, 2022-02-21:
@kbarilar unfortunately it doesn't solve the problem for me. I did a fresh clone of the project. With the native settings it throws the error. Then I added ```<string.h>``` in ```src/template.c```, completely removed ```build``` dir, ```cmake```'d and the same error.

@mikaelnousiainen thank you for your response! Yes, I thought about it, there are RS41ng.bin .hex and .map files. It is annoying. Probably there is some little difference between Fedora/Ubuntu in handling library paths that leads to the problem. I'll try it tonight or tomorrow on another Devuan machine to see if there's any change.

ManoDaSilva, 2022-08-15:
I had the same issue with Ubuntu 20.04 LTS; I just commented out 
`# add_subdirectory(tests)` 
In CMakeLists.txt

mikaelnousiainen, 2022-08-23:
Please use the new Docker environment for the build to avoid this issue. There are instructions in the README file on how to build the firmware using Docker.

closed 2022-08-23 by mikaelnousiainen