mikaelnousiainen/RS41ng #8: Will not decode on Kenwood TM-D710

Reid-n0rc, 2022-01-31:
I can get my FT-2DR to decode packets from the RS41ng, but not my Kenwood TM-D710. It was speculated that there could be a pre-emphasis/de-emphasis issue and could explain why one receiver works, while the other does not. 

Has pre/de-emphasis been accounted for?

mikaelnousiainen, 2022-02-18:
@Reid-n0rc I have a TM-D710G and it does successfully decode APRS, though it is indeed picky about the decoding. The AFSK encoding is really a hack on the sonde (see README for more details), as it can't really do it properly. There is no filtering/emphasis even possible with the hardware, it has not been designed to output AFSK at all.

mikaelnousiainen, 2022-02-20:
@Reid-n0rc There was actually a bug in the APRS call sign field generation for call signs shorter than 6 characters. This could have been the issue indeed if you used a short call sign like N0RC. Can you try to fetch the latest changes from Git and re-flash your sonde to see if it decodes now?

mikaelnousiainen, 2022-11-21:
Closing issue with no activity. I've had several reports about successful decodes using TM-D710(G) at least (including mine). However, the APRS modulation is far from perfect, as the hardware cannot properly modulate AFSK, so your mileage may vary.

closed 2022-11-21 by mikaelnousiainen