mikaelnousiainen/RS41ng #10: config.h should have APRSCALLSIGN in quotes ?

zanco, 2022-02-26:
Hi, I found out that #define APRS_CALLSIGN PE2BZ does not work, it actually requires "PE2BZ" which might be obvious for most of us, but took me some "debugging" (reading the error message)

mikaelnousiainen, 2022-02-26:
@zanco The default is to use a single call sign for all modes, so re-defining CALLSIGN in the beginning of the config file should be enough. That value has quotes. This is why APRS_CALLSIGN simply defines it as the same as CALLSIGN, which is how C language works. If you do not have any special case in question where you need multiple call signs, please simply define CALLSIGN and leave APRS_CALLSIGN to the default value of CALLSIGN.

So the idea is to leave APRS_CALLSIGN as it is.

mikaelnousiainen, 2022-02-26:
The main CALLSIGN value defaults to "MYCALL" where it should be hopefully obvious that it needs to be replaced :)

closed 2022-02-26 by mikaelnousiainen