mikaelnousiainen/RS41ng #9: What to expect if BMP280 is enabled but not connected ?

zanco, 2022-02-26:
Hi, today I did build the firmware on Ubuntu, .bin and .hex files are written, after re powering the sonde the red led stays on.

I have not yet attached the BMP280 but it's enabled in the config.

Does the software check and halt if no BMP280 is connected ?

Thanks,
Ben

mikaelnousiainen, 2022-02-26:
@zanco Also note that the code supports both BMP280 and BME280. With a BME280 you will get humidity readings, which are missing in a BMP280.

zanco, 2022-02-26:
Thanks for the info, after connecting the BME280 does the software find on which address it is, or do I have to provide an BME280 on either 0x76 or 0x77 ? (my values stay zero after repowering the RS41 with BME connected.

mikaelnousiainen, 2022-02-26:
The default is 0x77, but you can alter it here: https://github.com/mikaelnousiainen/RS41ng/blob/main/src/bmp280_handler.c#L9

mikaelnousiainen, 2022-02-26:
Also, we had some reports of 100 kHz I²C clock not working, so try reducing the clock to 10 kHz on this line (change 100000 to 10000):

https://github.com/mikaelnousiainen/RS41ng/blob/main/src/hal/i2c.c#L50

zanco, 2022-02-26:
/me is humming "under pressure" :-)

mikaelnousiainen, 2022-02-27:
@zanco I've committed more fixes to handling disconnected I²C devices like BMP280. There are also fixes to avoid transmitting invalid GPS position information, so please get the latest changes!

closed 2022-02-27 by mikaelnousiainen