mikaelnousiainen/RS41ng #14: Feature Request- 'Pips' between transmissions

darksidelemm, 2022-04-03:
RS41HUP had a useful feature where you could transmit short 'pips' (50ms of carrier) at regular intervals between telemetry transmissions. 
This acts as a channel marker of sorts, helping listeners tune their receivers in, while not using too much power.

I would suggest adding it in as a 'mode' which the user can include inamongst other transmissions.

e.g.
```
#define RADIO_POST_TRANSMIT_DELAY_MS 2000

#define RADIO_SI4032_TX_HORUS_V2 true
#define RADIO_SI4032_TX_HORUS_V2_COUNT 1
#define RADIO_SI4032_TX_PIP true
#define RADIO_SI4032_TX_PIP_COUNT 5
```

Would result in a Horus v2 transmission, then 6 pips every 2 seconds thereafter

mikaelnousiainen, 2022-04-07:
:+1: I will be implementing this soon (once I find the time to work on the firmware again :D). Should not be terribly difficult.

mikaelnousiainen, 2022-06-16:
@darksidelemm Added in commit https://github.com/mikaelnousiainen/RS41ng/commit/0aa49f5a2c156c74f0eb6c3ee8cdb8a7437719ea for both Si4032 and Si5351 transmitters

closed 2022-06-16 by mikaelnousiainen