mikaelnousiainen/RS41ng #15: make fails , ubuntu 18.04

zanco, 2022-04-07:
Hi, today I wanted to try the SI5351 as addon. Did apt update and upgrade on the Ubuntu 18.04 system. Removed the RS41ng directory and pulled a fresh one. Edited the config.h

cmake gives no errors. Make however starts showin something I completely don't understand. 

Up to 72 percent no problems, then a warning, then a lot of "problems" are reported, at least I guess they are problems, and make failed with error 2

Any idea what's going on here ? Thanks!

`[ 72%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/codecs/jtencode/lib/init_rs_int.cpp.o
/home/pe2bz/RS41ng/src/codecs/jtencode/lib/init_rs_int.cpp: In member function 'void* JTEncode::init_rs_int(int, int, int, int, int, int)':
/home/pe2bz/RS41ng/src/codecs/jtencode/lib/init_rs_int.cpp:33:29: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if(symsize < 0 || symsize > 8*sizeof(data_t)){
                     ~~~~~~~~^~~~~~~~~~~
[ 73%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/drivers/si5351/si5351.cpp.o
[ 75%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/drivers/si5351fast/si5351mcu.cpp.o
[ 76%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/si5351_handler.cpp.o
[ 77%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/si5351_test.cpp.o
[ 78%] Linking CXX executable RS41ng.elf
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(eh_arm.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(eh_globals.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(del_op.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(new_op.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(eh_personality.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(new_handler.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(eh_exception.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(eh_terminate.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(eh_unex_handler.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(class_type_info.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(eh_call.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(eh_term_handler.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(del_ops.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(pure.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(tinfo.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(eh_catch.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v7-m/libstdc++_nano.a(si_class_type_info.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: RS41ng.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: RS41ng.elf section `.co_stack' will not fit in region `ram'
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: region ram overflowed with stack
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 5300 bytes
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: region `ram' overflowed by 32 bytes
collect2: error: ld returned 1 exit status
src/CMakeFiles/RS41ng.elf.dir/build.make:1758: recipe for target 'src/RS41ng.elf' failed
make[2]: *** [src/RS41ng.elf] Error 1
CMakeFiles/Makefile2:86: recipe for target 'src/CMakeFiles/RS41ng.elf.dir/all' failed
make[1]: *** [src/CMakeFiles/RS41ng.elf.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
`

zanco, 2022-04-07:
I removed arm-none-eabi-gcc and had it installed again , making it now showing version 10.3-2021-10 when called with arm-none-eabi-gcc --version.

It still fails but with different message(s)

`[ 77%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/si5351_test.cpp.o
[ 78%] Linking CXX executable RS41ng.elf
collect2: fatal error: ld terminated with signal 7 [Bus error], core dumped
compilation terminated.
src/CMakeFiles/RS41ng.elf.dir/build.make:1758: recipe for target 'src/RS41ng.elf' failed
make[2]: *** [src/RS41ng.elf] Error 1
make[2]: *** Deleting file 'src/RS41ng.elf'
CMakeFiles/Makefile2:86: recipe for target 'src/CMakeFiles/RS41ng.elf.dir/all' failed
make[1]: *** [src/CMakeFiles/RS41ng.elf.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
`

signal 7 bus error ? Is that my Ubuntu system playing games with me ? 

Thanks again,
Ben

zanco, 2022-04-07:
So I did a clean 20.04 install on another machine. Installed by building instructiions gnu gcc toolchain and cmake

Still does not compile , after make this is the error where I am left with, and same error appears at a friend who runs this in Windwos WSL with 20.04 and the 8.3.1 version of arm-non-eabi

`[ 78%] Linking CXX executable RS41ng.elf
/usr/share/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: RS41ng.elf section `.co_stack' will not fit in region `ram'
/usr/share/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: region ram overflowed with stack
/usr/share/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: region `ram' overflowed by 24 bytes
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/RS41ng.elf.dir/build.make:1044: src/RS41ng.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:143: src/CMakeFiles/RS41ng.elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
`

mikaelnousiainen, 2022-04-07:
1) The Bus Error indicates some software/hardware issue that is not related to RS41ng.

2) The error "region ram overflowed by 24 bytes" indicates that the firmware does not fit into the sonde flash memory. You will need to disable some features (do you need all the modes?) or remove some of the defined text messages (e.g. for APRS) to reduce memory footprint.

mikaelnousiainen, 2022-08-23:
Please use the new Docker environment for the build to avoid this issue. There are instructions in the README file on how to build the firmware using Docker.

closed 2022-08-23 by mikaelnousiainen