mikaelnousiainen/RS41ng #18: out of ram and rom

zanco, 2022-08-02:
Hi, even with reducing / removing the APRS comments I can not find a setting which allows 4fsk on the RS41 radiochip and wspr on the SI5351 radiochip. 

Still running 156 rom short and 32 ram short.

Is there any way to reduce the memory used to get this combination working ?

Thanks,
Ben

micked99, 2022-08-02:
same here, error message:
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld: RS41ng.elf section `.ARM.exidx' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld: RS41ng.elf section `.co_stack' will not fit in region `ram'
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld: region ram overflowed with stack
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 88 bytes
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld: region `ram' overflowed by 144 bytes

mikaelnousiainen, 2022-08-02:
Please share your config.h files here, I can try to see what could be disabled to get more space.

micked99, 2022-08-03:
Tnx for getting back!
Here is my config https://pastebin.com/0CmSAztq
and
https://pastebin.com/4Kfk48sP

micked99, 2022-09-28:
I finally solved this by completely removing Horus V1 and some other small stuff from the fw, that freed up enough memory so I can flash it to run sequences of aprs/Horus V2 and wspr on both 70cm and on 20m with added 5351 card.
/Mikael

mikaelnousiainen, 2022-11-03:
@zanco @micked99 I'm closing this issue, as it can be fixed by disabling ALL unneeded features / transmission modes. It is unfortunate that when using Si5351, the code size grows a lot and some other modes will have to be disabled. Also the amount of text in message templates counts, so that can be reduced too!

closed 2022-11-03 by mikaelnousiainen