mikaelnousiainen/RS41ng #19: Callsigned specified isn't transmitted with Horus V2

ManoDaSilva, 2022-08-13:
Hi! 

First of all, I'd like to thank you for all the hard work! I've been playing around with your code, and we plan to use it with a ham weather balloon we're launching next month :) 

I'm currently having issues with the Horus V2 mode: no matter what callsign I specify in config.h, my RS-41 always transmits 4FSKTEST-V2 as payload ID. I've looked everywhere in the source files, but I cant' seem to find where it should be specified... 

Cheers,
Manoel

mikaelnousiainen, 2022-08-13:
You will need to request a new version 2 Horus 4FSK ID from https://github.com/projecthorus/horusdemodlib/issues by filing a new issue. It seems you already found the place though :)

And set it here:

https://github.com/mikaelnousiainen/RS41ng/blob/b0602f8b18c82c35ec04b4115c47dfb1d38b9164/src/config.h#L202

Horus 4FSK uses numeric IDs (and not any call sign directly).

mikaelnousiainen, 2022-08-13:
Also note that Horus 4FSK version 1 and version 2 use different set of IDs, so you'll need to specify V2 explicitly.

closed 2022-08-13 by mikaelnousiainen

ManoDaSilva, 2022-08-13:
Yup, I was just about to say I found it quite easily... Sorry for the bother! 

Quick question: I'm looking for an available pin I could use for an interrupt (as we want to add a geiger counter to our payload - already added a "driver" for it and implemented a custom value in the HorusV2 field). Do you think PB11 (broken out on the XDATA connector) could be a good candidate? I'm not going to use I2C/Serial anyway, and it seems TIM2_CH4 is available there as an alternate... Otherwise, I'm thinking about a free CS pin, but none of them have a timer available...

mikaelnousiainen, 2022-08-13:
@ManoDaSilva Actually, we have plans to implement such counter very soon, see discussion in issue #17