mikaelnousiainen/RS41ng #21: arm-none-eabi/bin/ld : Error: unable to disambiguate: -search_paths_first

fred-corp, 2022-08-21:
I'm trying to build the code on macOS; everything works well until I get to this point :

```
[ 78%] Linking CXX executable RS41ng.elf
/opt/local/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld : arm-none-eabi/bin/ld : Error: unable to disambiguate: -search_paths_first  (did you mean --search_paths_first ?)

make[2]: *** [src/RS41ng.elf] Error 1
make[1]: *** [src/CMakeFiles/RS41ng.elf.dir/all] Error 2
make: *** [all] Error 2
```

I had to specify my ```TOOLCHAIN_PATH``` in ```src/CMakeLists.txt``` to be ```/opt/local/bin``` before running ```cmake ..``` because the ```arm-none-eabi-gcc``` toolchain is apparently installed there on macOS.

How could I fix this issue ?

fred-corp, 2022-08-21:
Some progress :  
I installed the toolchain with HomeBrew instead of MacPorts (it is now in ```/usr/local/bin```).  
Now the error message shows something else : 

```
[ 78%] Linking CXX executable RS41ng.elf
arm-none-eabi-g++: error: RS41: No such file or directory
arm-none-eabi-g++: error: RS41ng/build/RS41ng.map: No such file or directory
make[2]: *** [src/RS41ng.elf] Error 1
make[1]: *** [src/CMakeFiles/RS41ng.elf.dir/all] Error 2
make: *** [all] Error 2
```

There seems to be a file missing, I'll try to investigate on that

fred-corp, 2022-08-21:
Apparently I needed ```libbsd-devel```, which I could not install on my machine... 

I built the program with the Dockerfile @MrARM made (linked to [this pull request](https://github.com/mikaelnousiainen/RS41ng/pull/11)), and it works great !

closed 2022-08-21 by fred-corp

mikaelnousiainen, 2022-08-21:
@fred-corp Ok, thanks for the report -- unfortunately I don't have a Mac to test the build.

Did the Dockerfile and its environment work for you or did you need to do some changes to it? What was the comment about libbsd-devel?

fred-corp, 2022-08-21:
@mikaelnousiainen 

## ```libbed-devel``` error

This is the error I get : 

```zsh
[ 78%] Linking CXX executable RS41ng.elf
/usr/local/Cellar/gcc-arm-none-eabi/20200630/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: warning: cannot find entry symbol arch_paths_first; defaulting to 0000000008000000
Building /Users/fredericdruppel/Documents/Projects/2022-08-01_RS41_Geiger/RS41ng/build/RS41ng.hex 
Building /Users/fredericdruppel/Documents/Projects/2022-08-01_RS41_Geiger/RS41ng/build/RS41ng.bin
   text    data     bss     dec     hex filename
  51004    1520    5160   57684    e154 RS41ng.elf
[ 78%] Built target RS41ng.elf
Scanning dependencies of target Project
[ 79%] Building C object tests/CMakeFiles/Project.dir/bell_test.c.o
/Users/fredericdruppel/Documents/Projects/2022-08-01_RS41_Geiger/RS41ng/tests/bell_test.c:16:54: warning: data argument not used by format string [-Wformat-extra-args]
    snprintf(test, sizeof(test), "%3$s %4$s %2$s\n", "1fd", gg, "3aa", "4ab");
                                 ~~~~~~~~~~~~~~~~~~  ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/usr/include/secure/_stdio.h:57:62: note: expanded from macro 'snprintf'
  __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
                                                             ^~~~~~~~~~~
/Users/fredericdruppel/Documents/Projects/2022-08-01_RS41_Geiger/RS41ng/tests/bell_test.c:27:13: warning: data argument not used by format string [-Wformat-extra-args]
            telemetry.locator,
            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/usr/include/secure/_stdio.h:57:62: note: expanded from macro 'snprintf'
  __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
                                                             ^~~~~~~~~~~
/Users/fredericdruppel/Documents/Projects/2022-08-01_RS41_Geiger/RS41ng/tests/bell_test.c:74:1: warning: non-void function does not return a value [-Wreturn-type]
}
^
3 warnings generated.
[ 80%] Building C object tests/CMakeFiles/Project.dir/morse_test.c.o
/Users/fredericdruppel/Documents/Projects/2022-08-01_RS41_Geiger/RS41ng/tests/morse_test.c:28:1: warning: non-void function does not return a value [-Wreturn-type]
}
^
1 warning generated.
[ 81%] Building C object tests/CMakeFiles/Project.dir/template_test.c.o
/Users/fredericdruppel/Documents/Projects/2022-08-01_RS41_Geiger/RS41ng/tests/template_test.c:3:10: fatal error: 'bsd/string.h' file not found
#include <bsd/string.h>
         ^~~~~~~~~~~~~~
1 error generated.
make[2]: *** [tests/CMakeFiles/Project.dir/template_test.c.o] Error 1
make[1]: *** [tests/CMakeFiles/Project.dir/all] Error 2
make: *** [all] Error 2
```

the ```.elf``` file gets made, but I'm missing ```bad/string.h``` which I figured is part of libbsd-devel.

## Docker build environment

I just changed the following lines in ```docker_build.sh``` from this : 

```zsh
# Copy the binary, ELF and hex to the home directory
cp src/RS41ng.elf /usr/src/rs41ng
cp RS41ng.hex /usr/src/rs41ng
cp RS41ng.bin /usr/src/rs41ng
```

to this : 

```zsh
# Copy the binary, ELF and hex to the home directory
rm -rf /usr/src/rs41ng/build/
mkdir /usr/src/rs41ng/build/
cp src/RS41ng.elf /usr/src/rs41ng/build/
cp RS41ng.hex /usr/src/rs41ng/build/
cp RS41ng.bin /usr/src/rs41ng/build/
```

So the container automatically replaces the old build files with new ones in a ```build```directory.  

Then I just ran the following commands to create the container and build the code (as mentioned in @MrARM 's README) : 

```zsh
docker build -t rs41ng_compiler .
docker run --rm -it -v $(pwd):/usr/src/rs41ng rs41ng_compiler
```

And it works a treat !