mikaelnousiainen/RS41ng #22: Error 2 when docker compiling

MrTalon63, 2022-09-04:
It looks like when make tries to create a .elf file it has some unknown problems
```[ 97%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/si5351_handler.cpp.o
[ 98%] Building CXX object src/CMakeFiles/RS41ng.elf.dir/si5351_test.cpp.o
/usr/local/src/RS41ng/src/codecs/jtencode/lib/init_rs_int.cpp: In member function 'void* JTEncode::init_rs_int(int, int, int, int, int, int)':
/usr/local/src/RS41ng/src/codecs/jtencode/lib/init_rs_int.cpp:33:29: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
   33 |   if(symsize < 0 || symsize > 8*sizeof(data_t)){
      |                     ~~~~~~~~^~~~~~~~~~~~~~~~~~
[100%] Linking CXX executable RS41ng.elf
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/RS41ng.elf.dir/main.c.o: in function `main':
/usr/local/src/RS41ng/src/main.c:74: multiple definition of `main'; CMakeFiles/RS41ng.elf.dir/build/CMakeFiles/3.21.5/CompilerIdC/CMakeCCompilerId.c.o:/usr/local/src/RS41ng/src/build/CMakeFiles/3.21.5/CompilerIdC/CMakeCCompilerId.c:767: first defined here
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/RS41ng.elf.dir/build/CMakeFiles/3.21.5/CompilerIdCXX/CMakeCXXCompilerId.cpp.o: in function `main':
/usr/local/src/RS41ng/src/build/CMakeFiles/3.21.5/CompilerIdCXX/CMakeCXXCompilerId.cpp:757: multiple definition of `main'; CMakeFiles/RS41ng.elf.dir/build/CMakeFiles/3.21.5/CompilerIdC/CMakeCCompilerId.c.o:/usr/local/src/RS41ng/src/build/CMakeFiles/3.21.5/CompilerIdC/CMakeCCompilerId.c:767: first defined here
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/RS41ng.elf.dir/build/CMakeFiles/3.21.5/CompilerIdCXX/CMakeCXXCompilerId.cpp.o:/usr/local/src/RS41ng/src/build/CMakeFiles/3.21.5/CompilerIdCXX/CMakeCXXCompilerId.cpp:738: multiple definition of `info_language_dialect_default'; CMakeFiles/RS41ng.elf.dir/build/CMakeFiles/3.21.5/CompilerIdC/CMakeCCompilerId.c.o:/usr/local/src/RS41ng/src/build/CMakeFiles/3.21.5/CompilerIdC/CMakeCCompilerId.c:754: first defined here
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/RS41ng.elf.dir/build/CMakeFiles/3.21.5/CompilerIdCXX/CMakeCXXCompilerId.cpp.o:/usr/local/src/RS41ng/src/build/CMakeFiles/3.21.5/CompilerIdCXX/CMakeCXXCompilerId.cpp:718: multiple definition of `info_arch'; CMakeFiles/RS41ng.elf.dir/build/CMakeFiles/3.21.5/CompilerIdC/CMakeCCompilerId.c.o:/usr/local/src/RS41ng/src/build/CMakeFiles/3.21.5/CompilerIdC/CMakeCCompilerId.c:733: first defined here
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/RS41ng.elf.dir/build/CMakeFiles/3.21.5/CompilerIdCXX/CMakeCXXCompilerId.cpp.o:/usr/local/src/RS41ng/src/build/CMakeFiles/3.21.5/CompilerIdCXX/CMakeCXXCompilerId.cpp:717: multiple definition of `info_platform'; CMakeFiles/RS41ng.elf.dir/build/CMakeFiles/3.21.5/CompilerIdC/CMakeCCompilerId.c.o:/usr/local/src/RS41ng/src/build/CMakeFiles/3.21.5/CompilerIdC/CMakeCCompilerId.c:732: first defined here
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/RS41ng.elf.dir/build/CMakeFiles/3.21.5/CompilerIdCXX/CMakeCXXCompilerId.cpp.o:/usr/local/src/RS41ng/src/build/CMakeFiles/3.21.5/CompilerIdCXX/CMakeCXXCompilerId.cpp:388: multiple definition of `info_compiler'; CMakeFiles/RS41ng.elf.dir/build/CMakeFiles/3.21.5/CompilerIdC/CMakeCCompilerId.c.o:/usr/local/src/RS41ng/src/build/CMakeFiles/3.21.5/CompilerIdC/CMakeCCompilerId.c:403: first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/RS41ng.elf.dir/build.make:1153: src/RS41ng.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:117: src/CMakeFiles/RS41ng.elf.dir/all] Error 2
make: *** [Makefile:91: all] Error 2```

mikaelnousiainen, 2022-11-03:
@MrTalon63 Have you tried compiling using Docker (see instructions in the README file) and possibly using the workaround in #23 ?

mikaelnousiainen, 2022-11-21:
Closing issue with no activity.

closed 2022-11-21 by mikaelnousiainen