mikaelnousiainen/RS41ng #25: Can you try if FT8 still is working ?

zanco, 2022-10-06:
Hi, I can compile a version which transmits wspr , Horus V2 and FT8 on HF (si5351) but the FT8 signal however is undecodable.

Instead of a normal FT8 transmission there appears an FT8 signal which starts at the right moment, sounds normal, but got interrupted, restarts, got interrupted, restarts and ends by the usual timing

[image]

The first transmission is the FT8 transmission, followed by the 4FSK transmission. As you can see the FT8 is "interrupted" 

The same problem happened with another user with another computer, other RS41 and si5351 too.

I would like to test this mode on a balloon flight :-)

Thanks,

Ben

mikaelnousiainen, 2022-11-03:
@zanco Can you share your `config.h` file here? I can try to test FT8, but I need your configuration for it!

zanco, 2022-11-04:
[bz-config.h.txt](https://github.com/mikaelnousiainen/RS41ng/files/9938315/bz-config.h.txt)
Sure,

Has V2 on SI4032 enabled, has V2 and FT8 on SI5351 enabled. FT8 transmits but as shown on the waterfall it cannot be decoded.
Thanks,
Ben

micked99, 2022-11-05:
I have the same problem with the latest fw, I downloaded an older version of the fw an there the ft8 is working well 

https://github.com/mikaelnousiainen/RS41ng/tree/d12c923594ef80f896b4b07db6b51266919498b3

mikaelnousiainen, 2022-11-21:
@zanco @micked99 Thanks for the reports, I'd have to get back to testing this functionality, but have been busy on other things lately.

Please have a look at how you can enable debug logging (using semilogging) here: https://github.com/mikaelnousiainen/RS41ng/#developing--debugging-the-firmware -- this will provide you information about time sync and exact transmission decisions!

Posting some FT8 transmission logs here would help greatly!

Also, are you sure the sonde has a GPS lock? Otherwise timing will be off.

mikaelnousiainen, 2023-08-09:
According to recent reports, FT8 is working well. Please pull the latest code and test. This issue can be reopened if the problem persists.