mikaelnousiainen/RS41ng #32: GPS power saving

albertus62, 2022-12-03:
Please add GPS power saving as used in RS41HUP (UBLOX_POWERSAVE). The batteries last much longer needed for floating HAB's.

darksidelemm, 2023-05-22:
Just as some references:
```
// Power Consumption Notes (all @ 3V):
// - GPS Max Performance, Transmitting @ 13 dBm = ~150 mA
// - GPS Max Performance, Not Transmitting = 70-90mA
// - GPS in PowerSave Mode, Transmitting @ 13 dBm = ~120 mA
// - GPS in PowerSave Mode, not Transmitting = 30-50mA, depending on GPS state.
```

Power-Save mode setting in RS41HUP: https://github.com/darksidelemm/RS41HUP/blob/master/ublox.c#L73
Worth noting that I only entered powersave mode once the GPS had a valid fix, otherwise it could take considerably longer to get initial lock: https://github.com/darksidelemm/RS41HUP/blob/master/main.c#L482

From what I recall the powersave mode worked fairly well. It was when i tried to shut the GPS down between transmissions where things went a bit iffy, sometimes resulting in the GPS never getting lock.

darksidelemm, 2023-08-08:
I found that stopping / starting the GPS was a bit unreliable. The GPS would sometimes take a very very long time to get lock on restart, or not get lock at all.

mikaelnousiainen, 2023-08-09:
@albertus62 @YT3GTI @whallmann I will attempt to implement a simple approach to enable GPS power saving this week  (the same approach as in RS41HUP to turn on power saving mode -- but NOT stop the GPS chip completely -- after a successful fix has been acquired)

mikaelnousiainen, 2023-08-09:
@albertus62 @YT3GTI @whallmann @darksidelemm Simple GPS power saving (the RS41HUP approach) is implemented now in https://github.com/mikaelnousiainen/RS41ng/commit/fda9cef76a3dd9cf48288f254d3d0e638ad6550a

This is now totally untested (and disabled by default). Please test :)

darksidelemm, 2023-08-10:
Thanks for this! 

However, when power save mode is inactive, you shouldn't be modifying the sats field ( https://github.com/mikaelnousiainen/RS41ng/blob/fda9cef76a3dd9cf48288f254d3d0e638ad6550a/src/codecs/horus/horus_packet_v2.c#L46 )

The other conversions (+100 for tracking state, and +200 for low power optimized state) are correct.