mikaelnousiainen/RS41ng #33: APRS Latitude not showing

currentusername, 2022-12-09:
My latitude is 150.... and does not show up on my FT3D is that because $lat is up to 6 characters and my latitude is 7?

currentusername, 2022-12-10:
Just tried $lat $lon in the APRS comment and they both appear there to 6 characters, but in the APRS telemetry data part of the message it only shows the latitude.

mikaelnousiainen, 2022-12-18:
Thanks for reporting this. I'll have a look at the issue soon.

neilogd, 2022-12-28:
Seeing similar too, direwolf output in case it's of use:

VK2GPU-11 audio level = 16(10/10)   [NONE]   ___|||___
[0.4] VK2GPU-11>APZ41N,WIDE1-1,WIDE2-1:!3343.56S/105233.207WO086/000/A=000583/P3S6T37V3956C00 Bat: 3956 mV, Sats: 6,  RS41ng FOX
Unexpected character "3" found where period expected in longitude.
Position, TBD, Experimental
S 33 43.5600, Invalid Longitude, alt 583 ft
WO086/000/P3S6T37V3956C00 Bat: 3956 mV, Sats: 6,  RS41ng FOX

mikaelnousiainen, 2023-08-09:
@currentusername @neilogd It's been a long time, but I finally think I figured out what the issue was. The code (that I copied over from some older RS41 firmware codebase) was using 8-bit signed values for degrees, so of course it did not support longitudes > 128 or < -127. Now this should be fixed in the codebase, please give it a try!

currentusername, 2023-08-11:
Just tried it out and it works! Thank you!

mikaelnousiainen, 2023-08-12:
Great, thanks for testing, closing the issue now.

closed 2023-08-12 by mikaelnousiainen