mikaelnousiainen/RS41ng #35: Run command create $(pwd) invalid characters in Docker Desktop for Windows

dalewegner, 2023-02-20:
The READ.md file instruction Build command worked fine.  But the run command generated this error.  I tried different direct formats for pwd but could not seem to get it to work.  Running Windows 10 with Docker Desktop for Windows.  Any idea what syntax is creating this error?

 O:\My Documents\Balloon\Sonde\RS41ng Reprogramming\RS41ng-main>docker run --rm -it -v $(pwd):/usr/local/src/RS41ng rs41ng_compiler
docker: Error response from daemon: create $(pwd): "$(pwd)" includes invalid characters for a local volume name, only "[a-zA-Z0-9][a-zA-Z0-9_.-]" are allowed. If you intended to pass a host directory, use absolute path.
See 'docker run --help'.

dalewegner, 2023-02-20:
Seems like I cannot use a network drive and have to use a direct path to get it to work.  This is command that worked.

c:\Balloon\RS41ng-main>docker run --rm -it -v "c:\Balloon\RS41ng-main":/usr/local/src/RS41ng rs41ng_compiler

Hate it when I answer my own question.

Arm-D, 2023-08-08:
In Windows command line tool (CMD) the expression with $(pwd) does not work. 
The correct expression to mount the drive is: %cd%
You must use: docker run --rm -it -v %cd%:/usr/local/src/RS41ng rs41ng_compiler

mikaelnousiainen, 2023-08-09:
Thanks!

@dalewegner RS41ng really only supports use of Windows Subsystem for Linux (WSL) and the `bash` shell for builds.

@Arm-D But if you confirm that the command you wrote down works, I can add it to the documentation. I would still advise to use WSL for builds on Windows too!

Arm-D, 2023-08-09:
I do not know WSL, so I just use the Windows command line tool.
To compile the code I followed exactly the steps in the read.me file. 
The only difference is the use of %cd% instead of $(pwd).
I confirm this works fine, so please add it to the documentation, maybe it helps other users.

iz2fly, 2023-08-09:
@Arm-D I'm no expert, my son taught me.
I use windows **PowerShell** and this works.
docker run --rm -it -v ${pwd}:/usr/local/src/RS41ng rs41ng_compiler

mikaelnousiainen, 2023-08-09:
I added the Windows Command Prompt version to the README.

closed 2023-08-09 by mikaelnousiainen