mikaelnousiainen/RS41ng #38: Flashing instructions are unclear

argilo, 2023-06-01:
The flashing instructions in the readme are:

> 1. Remove batteries from the sonde
> 2. Connect an ST-LINK v2 programmer dongle [...]
> 3. Unlock the flash protection [...]

These instructions didn't work for me. Instead I had to put the batteries in the sonde and press the power button to switch it on before unlocking & flashing would work. The ST-LINK v2 does not appear to supply power to the target device (but it could be that I connected it incorrectly.)

If the instructions need updating, I'd be happy to open a pull request.

pinkavaj, 2023-06-02:
The unlocking/flashing instructions worked for me, no batteries were necessary. I'm using some cheap ST-LINK v2 clone.

What comes to my mind:
* broken wire
* your programmer have power mamangement and the output needs to be explicitly enabled
* somehow broken powesupply in your programmer
Pls let us know if you find out.

argilo, 2023-06-02:
I'm using a genuine ST-LINK v2. The manual is not entirely clear, but I don't believe it is intended to supply power to the target device.

iz2fly, 2023-06-02:
[image]
I use this connection without batteries and it works fine.
I also happened to find an ST-LinkV2 with the wrong pin descriptions, the ground was in the wrong place and I burned it.

iz2fly, 2023-06-02:
I use this connection without batteries and it works fine. There are different versions.
I also happened to find an ST-LinkV2 with the wrong pin descriptions, the ground was in the wrong place and I burned it.

[image]

pinkavaj, 2023-06-02:
The schemas seems to be identical, I guess the other should use the 3.3V ?

mikaelnousiainen, 2023-07-03:
@argilo @pinkavaj @iz2fly 

There are indeed different types of ST-LINK adapters and I wrote the instructions based on the small USB dongle type adapter I have. Some adapters might behave differently.

It is indeed possible to flash the sonde by powering it from batteries and NOT connecting the +3.3V from the ST-LINK adapter at all -> the only three wires needed with batteries are the GND, SWDIO and SWCLK.

argilo, 2023-07-17:
Thanks for the clarification. I opened a pull request to update the README with that information: #48

PR #48 (2023-08-08): Add a clarification about powering the sonde
Fixes #38.
commit 72a33f13 (2023-08-08): Add a clarification about powering the sonde (#48)

Fixes #38
closed 2023-08-08 by mikaelnousiainen