mikaelnousiainen/RS41ng #39: Looking for some explanation about the  "TIME_SYNC"

zanco, 2023-06-03:
Hi, I am trying to make RS41ng running 12 payloads, with different ID's on the same frequency (intended for a scouting competition)

I don't have enabled continous for either V1 or V2 4fsk, all other modes are disabled.

`// Use Horus payload ID 256 for Horus V2 tests (4FSKTEST-V2)
#define HORUS_V2_PAYLOAD_ID 379
#define HORUS_V2_BAUD_RATE_SI4032 100
#define HORUS_V2_BAUD_RATE_SI5351 100
#define HORUS_V2_PREAMBLE_LENGTH 16
#define HORUS_V2_IDLE_PREAMBLE_LENGTH 32
#define HORUS_V2_TONE_SPACING_HZ_SI5351 270
#define HORUS_V2_TIME_SYNC_SECONDS 10         // Schedule transmission every N seconds, counting from beginning of an hour (based on GPS$
#define HORUS_V2_TIME_SYNC_OFFSET_SECONDS 60  // Delay transmission for an N second offset after the scheduled time.`

With these settings I expected the sonde to only transmit every 10 seconds AND again after the 60 seconds offset. 

I added another tracker with the V2_TIME_SYNC_SECONDS set to 5 but both trackers, with GPS lock, appear to transmit continuously.

Can you shine a light on my question and maybe explain where I am going wrong ? Thanks.

mikaelnousiainen, 2023-07-03:
@zanco 

The time sync feature is a very simplistic way to activate the transmissions every N seconds, delayed by the "offset" setting.

Based on what you describe (using 12 payloads), you MIGHT be able to make this work by setting a constant TIME_SYNC_SECONDS to the full interval where all 12 payloads are expected to finish transmissions (do take into account 4FSK transmission length too -- though it's quite short).

For example with 3 payloads:

Payload 1:

```
#define HORUS_V2_TIME_SYNC_SECONDS 120 // everything happens at 120 seconds interval
#define HORUS_V2_TIME_SYNC_OFFSET_SECONDS 0 // the first payload will transmit at 0 seconds within the 120 second interval
```

Payload 2:

```
#define HORUS_V2_TIME_SYNC_SECONDS 120 // everything happens at 120 seconds interval
#define HORUS_V2_TIME_SYNC_OFFSET_SECONDS 40 // the second payload will transmit at 40 seconds within the 120 second interval
```

Payload 3:

```
#define HORUS_V2_TIME_SYNC_SECONDS 120 // everything happens at 120 seconds interval
#define HORUS_V2_TIME_SYNC_OFFSET_SECONDS 80 // the third payload will transmit at 80 seconds within the 120 second interval
```

Also please note that the current code does not take into account leap seconds -- though that doesn't matter with 4FSK transmissions (but only with time-slotted ones like WSPR or FT8).

zanco, 2023-07-23:
Hi, thanks for your reply, now testing with 2 trackers, one with offset 20 and one with offset 30, and 120 seconds in interval.

Withing reasonable deviation ( 9 - 11 seconds) they stay apart from each other, but at 00 and 10 offset. Does not matter to much, will add another 2 tomorrow with 40 and 50 offset to check if they match my expected pattern :-)

mikaelnousiainen, 2023-08-09:
Now there is support for (a hard-coded) leap seconds setting, so it can be removed from the OFFSET. Please pull the latest changes. I'll add this explanation to README for others. Closing this issue now.

closed 2023-08-09 by mikaelnousiainen

zanco, 2023-08-17:
Hardcoded leap seconds works as a charm, thanks !!