mikaelnousiainen/RS41ng #41: Radiosonde not detected by RTL-SDR

prasadnoolu, 2023-06-17:
Hi, i have RTL-SDR and SDR++ application to see the responce from the radiosonde, i have programmed RS41ng firmware into my radiosonde and i tried to tune my sdr++ to below frequency, but no luck
 **// Transmit frequencies for the Si4032 transmitter modes
#define RADIO_SI4032_TX_FREQUENCY_CW        432500000
#define RADIO_SI4032_TX_FREQUENCY_PIP       432500000
#define RADIO_SI4032_TX_FREQUENCY_APRS_1200 432500000**
My SDR is not detecting any signal in the programmed frequency range, can anybody help me on this.

iz2fly, 2023-06-21:
Try only one mode first, say CW only and see if that works. I don't know if it supports multi-mode transmission.

prasadnoolu, 2023-06-21:
> config.zip
> Try this config.h, transmit CW at 432.500MHz

Hi I am able to resolve the issue by changing some settings in the Hours
and SDR++.

Thanks for your help.
Prasad

mikaelnousiainen, 2023-07-03:
Ok, sounds like it was an issue in SDR++ settings. Closing.

closed 2023-07-03 by mikaelnousiainen