mikaelnousiainen/RS41ng #50: Supported Sensors

gomes89, 2023-08-07:
Hello. Is it possible use the original temperature/humidity sensors in the boom? If so, how to configure the code to use them?

Also, same question applies to the barometer module in the SGP model.

iz2fly, 2023-08-07:
With this firmware I think it's not possible.
It is possible to detect, for now, only the temperature with the firmware you find on my website, modified by my friend.
https://iz2fly.altervista.org/radiosonde/
The calibration instructions are only in Italian and Spanish but I don't think it's a problem, google translate helps us.
I am using this firmware as CW beacon at 432.428MHz and on the APRS map (IZ2FLY-1), there you can see the temperature at 1150m above sea level with the original sensor in the boom:
https://aprs.fi/#!lat=45.6&lng=10.33083
Unfortunately there are few iGates on this frequency and they are not always in operation.

gomes89, 2023-08-07:
Thanks a lot for the reply @iz2fly. Indeed I didn't find any reference in the code. I will have a look at the firmware in your website and do a couple tests. Spanish or Italian are both fine for me without translation.

One thing I will have to check is the status of the boom in the radiosonde I plan to use. The original firmware reports PTU failure, but physically I see no damage at all (with microscope). With the pinout you provide in the calibration instructions I can easily measure the resistance.

Also, do you have any way to read the barometer in the RS41-SGP? With the original firmware I'm getting the correct value right now.

iz2fly, 2023-08-08:
Hi gomes89!
Right now I don't have a way to read the barometer with the firmware and humidity of my friend I2NOS. Maybe in the future it could be possible as we did with the temperature sensor.

mikaelnousiainen, 2023-08-08:
Thanks for the discussion here. I'm re-opening this to keep the discussion going on in case we arrive at some solution.

RS41ng does not (yet) support reading the temperature or humidity values using the built-in sensor stalk, because it is not known (not reverse-engineered) entirely how to use the sensors. I know there have been some attempts to reverse-engineer use of the sensors. @iz2fly Would you be able to provide the code that you use for reading the temperature?

iz2fly, 2023-08-08:
The code can be downloaded from the link on my web site that I put above:
https://iz2fly.altervista.org/radiosonde/

gomes89, 2023-08-08:
> Thanks for the discussion here. I'm re-opening this to keep the discussion going on in case we arrive at some solution.
>
> RS41ng does not (yet) support reading the temperature or humidity values [...]

Sure, just wanted to avoid discussing this here since it is more a feature request than an issue. I'm working myself on the barometer reading, will get back here if I reach any conclusion.

darksidelemm, 2023-08-10:
I know that Gene @wb9coy has also written code to read the temperature sensors, and the pressure sensor in the RS41-SGP. Hopefully they will chime in here with some information.

mikaelnousiainen, 2023-08-18:
@wb9coy Thanks for all the reverse-engineering work! We'll definitely want to incorporate both the pressure sensor code and any upcoming temperature sensor support in RS41ng.

I didn't see any license in RS41cam code: how is your code licensed?

Also, does your (upcoming temp sensor) code support reading humidity?

73,
Mikaem OH3BHX

wb9coy, 2023-08-18:
This is my original art and NOT copied from ANY other source.  It is open to all for non-profit.  I really should put a licensed statement in the code.

This is a youtube about the camera feature  https://www.youtube.com/watch?v=HijQeXucqZ0

mikaelnousiainen, 2023-08-18:
@wb9coy Yes, this is exactly what I mean: please indicate a suitable license in your repository, preferably something that is compatible with GPL-2.0 which is the license RS41ng uses :)

wb9coy, 2023-09-12:
I added the LICENSE file to the archive.

ECVGG, 2024-02-11:
> Hi Mark I am still packaging up the code for the temperature sensor. We flew a beta version a few weeks ago and I was pleased with the temperature readings that we observed. The RPM411 pressure sensor code is working perfectly. The build that has the pressure sensor working is at https://github.com/wb9coy/RS41cam

Could you make some steps for programming it? I'm a bit lost