mikaelnousiainen/RS41ng #52: HORUS and APRS with custom time schedule

9A4AM, 2023-08-24:
Hi!
I trying setup for two modes. HORUS and APRS with custom time schedule. APRS 300 seconds (5 minute) and HORUS 60 seconds (1 minute). I use APRS_TIME_SYNC_SECONDS 300 and HORUS_V2_TIME_SYNC_SECONDS 60 in Config.h, but APRS send packet every 300 seconds and HORUS send packet 60 seconds after APRS packet. If it possible set time for APRS send every 300 seconds and HORUS every 60 second, if yes, how??
Thanks in advance.
Mario.

mikaelnousiainen, 2023-08-29:
@9A4AM The time sync features are very primitive so the way you're configuring time sync leads to the behavior you describe -- it's all expected.

The way you can achieve what you want is to set BOTH APRS and HORUS_V2 time sync to 60 (seconds). This way all transmissions happen at the start of a minute. To repeat Horus 4 times (so every 5th will be APRS), you should set `RADIO_SI4032_TX_HORUS_V2_COUNT` to 4 and `RADIO_SI4032_TX_APRS_COUNT` to 1.

mikaelnousiainen, 2023-08-29:
Closing, as this is not really an issue. Please indicate if this works for you @9A4AM

9A4AM, 2023-08-29:
It's work, thanks.