mikaelnousiainen/RS41ng #58: APRS_Destination & APRS_Relays ??

K4KDR, 2023-09-23:
With the arrival of my first RS-41, I finally have a test platform to start learning about your wonderful software!  Thanks in advance for making this available.

Everything worked great on my first attempt to reflash the RS-41 and both the **4FSK** (via the Horus-GUI) and **APRS** (via Direwolf) are decoding perfectly.

My questions are:

1) is **APRS_DESTINATION** set to "APZ41N" for a particular reason?  Would you prefer that we leave it set to that value?

2) I thought that for high-altitude transmitters like balloons, using digi paths such as **WIDE1-1,WIDE2-1** was highly discouraged.   Would it be ok for me to change that line to:
`#define APRS_RELAYS ""`

... or would that cause some problem?

Thanks VERY much for help on these questions when time permits.

-Scott,  K4KDR

mikaelnousiainen, 2023-09-23:
@K4KDR Thanks. It is true that using digipeated WIDE paths could be problematic, so feel free to use no path at all. However, in most places the 70cm APRS network is either non-existent or with very few digipeaters, so this path will only increase coverage. This is why I left the default that way. On 2 meters (where the RS41 cannot even transmit) this path could be more of an issue.

The APZ41N just indicates the software that sends the packet. The APZ prefix indicates experimental software. Please have a look at APRS docs if you wish to know more.

mikaelnousiainen, 2023-09-23:
Btw I have not tested using no path at all, so please first verify it works.

K4KDR, 2023-09-23:
Many thanks for all that info!

While I am very much in experimental / learning mode, I did setup a 70cm Direwolf APRS iGate to see how my RS-41 beacon might look.  No idea what the final format might be, but as a first test, it showed up on APRS.fi just fine.  Thanks again for the helpful reply!

[image]

mikaelnousiainen, 2023-09-23:
:+1: Closing this as it's a question, not an issue. Please see the Discussions tab https://github.com/mikaelnousiainen/RS41ng/discussions for this type of questions or when asking for help!

closed 2023-09-23 by mikaelnousiainen