mikaelnousiainen/RS41ng #68: Cant get Si5351 running - need a sample config.h please

whallmann, 2023-11-04:
Is there a sample config.h where the si5351 board is running? I conncected a board to pin header of RS41 Ports 2 SDA and 3 SCL - see 3,8 volt on the VCC on si5351 - also set Si5351 CW on but see no signal on the clock0 output
[image]
.  my config.h is appended.
Any help is welcome.
Wolf - DF7PN
[df7pn_si5351_cw_config_h.txt](https://github.com/mikaelnousiainen/RS41ng/files/13256390/df7pn_si5351_cw_config_h.txt)

whallmann, 2023-11-04:
i debuged a while and see that the board gets init ok. Seems there is Something wrong with the board. Good that i bought 2 of them. will check tomorrow.

whallmann, 2023-11-07:
Meanwhile i connect another board. I set TX on 14 MHz and see a wide 500 kHz Signal in the Waterfall sounds crazy and waving around on the band. I tried power down to minimum 0 dB but nothing changes. So this project gets stopped now.

mikaelnousiainen, 2023-11-08:
@whallmann Sounds definitely odd. Are you sure the voltage from your modded power supply (with just 1xAA battery) stays stable? I have only tested it (very successfully) with 2xAAs and no power supply mod.

The config entries that matter -- the following should send CW at 3595 kHz:

```
// Enable use of an externally connected I²C Si5351 clock generator chip for HF radio transmissions
#define RADIO_SI5351_ENABLE true

// Si5351 transmit power: 0..3
// Si5351 drive strength: 0 = 2mA, 1 = 4mA, 2 = 6mA, 3 = 8mA
#define RADIO_SI5351_TX_POWER 3

// Which modes to transmit using an externally connected Si5351 chip in the I²C bus
// The COUNT settings define the number of times that each type of transmission is repeated
#define RADIO_SI5351_TX_CW true
#define RADIO_SI5351_TX_CW_COUNT 1
#define RADIO_SI5351_TX_PIP false
#define RADIO_SI5351_TX_PIP_COUNT 6
#define RADIO_SI5351_TX_HORUS_V1 false
#define RADIO_SI5351_TX_HORUS_V1_COUNT 1
#define RADIO_SI5351_TX_HORUS_V2 false
#define RADIO_SI5351_TX_HORUS_V2_COUNT 4
#define RADIO_SI5351_TX_JT9 false
#define RADIO_SI5351_TX_JT9_COUNT 1
#define RADIO_SI5351_TX_JT65 false
#define RADIO_SI5351_TX_JT65_COUNT 1
#define RADIO_SI5351_TX_JT4 false
#define RADIO_SI5351_TX_JT4_COUNT 1
#define RADIO_SI5351_TX_WSPR false
#define RADIO_SI5351_TX_WSPR_COUNT 1
#define RADIO_SI5351_TX_FSQ false
#define RADIO_SI5351_TX_FSQ_COUNT 1
#define RADIO_SI5351_TX_FT8 false
#define RADIO_SI5351_TX_FT8_COUNT 1

// Transmit frequencies for the Si5351 transmitter modes
#define RADIO_SI5351_TX_FREQUENCY_CW         3595000UL
#define RADIO_SI5351_TX_FREQUENCY_PIP        3595000UL
#define RADIO_SI5351_TX_FREQUENCY_HORUS_V1   3608000UL
#define RADIO_SI5351_TX_FREQUENCY_HORUS_V2   3608000UL
#define RADIO_SI5351_TX_FREQUENCY_JT9        14085000UL    // Was: 14078700UL
#define RADIO_SI5351_TX_FREQUENCY_JT65       14085000UL    // Was: 14078300UL
#define RADIO_SI5351_TX_FREQUENCY_JT4        14085000UL    // Was: 14078500UL
#define RADIO_SI5351_TX_FREQUENCY_WSPR       14085000UL    // Was: 14097200UL
#define RADIO_SI5351_TX_FREQUENCY_FSQ        3608350UL    // Was: 7105350UL     // Base freq is 1350 Hz higher than dial freq in USB
#define RADIO_SI5351_TX_FREQUENCY_FT8        14085000UL    // Was: 14075000UL
```

Also remember to set the CW message in `config.c` (this would send out the call sign):

```
char *cw_message_templates[] = {
        "$cs",
        NULL
};

```

That said, I'd need to test this again locally if there is some regression...

Can you share your config.h and config.c files -- put them as attachments here in the comments (click the attach file icon above).

Also, can you take a photo of the waterfall spectrum or maybe record a video? Having something 500 kHz wide sounds like the on-board crystal on the Si5351 break-out board would not be working properly at all.

I've personally used the same type of Si5351 board (as in the photo).

iz2fly, 2023-11-28:
At how many volts is the Si5351 powered?
Why is the RS41 powered only by a battery?
Why is the RS41 PCB broken? This could be the cause of the error.

GianlucaB2000, 2024-01-18:
Where can I find the diagram to connect the si5351 to the ? pins

Roger69, 2026-08-28:
#define RADIO_SI4032_TX_CW false

Turned off!!!!

whallmann, 2026-08-28:
> #define RADIO_SI4032_TX_CW false
> 
> Turned off!!!!

what you mean with this?

Roger69, 2026-08-28:
In the config.h file above , that line turns off the SI4032 radio.
for CW.