mikaelnousiainen/RS41ng #69: Building docker compiler

rpratt20, 2023-11-07:
I am building on an old Dell 64 bit running Ubuntu 20.04 Jammy jellyfish. Have attemped 2 times with errors :

randy@randy-Inspiron-N4010:~/RS41ng-main$ docker build -t rs41ng_compiler .
[+] Building 2775.4s (5/8)                                       docker:default
[+] Building 3055.4s (6/8)                                       docker:default
 => [internal] load build definition from Dockerfile                       1.9s
 => => transferring dockerfile: 312B                                       0.4s
 => [internal] load .dockerignore                                          1.6s
 => => transferring context: 2B                                            0.1s
 => [internal] load metadata for docker.io/library/fedora:36               3.2s
 => [1/4] FROM docker.io/library/fedora:36@sha256:64cd00a0e2b92d527c0a  1112.2s
 => => resolve docker.io/library/fedora:36@sha256:64cd00a0e2b92d527c0a095  0.4s
 => => sha256:64cd00a0e2b92d527c0a0954162a73e85f160e3a53c3832 975B / 975B  0.0s
 => => sha256:c3928ce4e65d6d1e66fc1e2c9b1cc163d31bc63a96117d9 529B / 529B  0.0s
 => => sha256:1d81a0524f7d02b7673be0eb1201ab507468b736188 1.99kB / 1.99kB  0.0s
 => => sha256:c7bef7d09442a4c4deef8e8120027e7fa3f163 60.07MB / 60.07MB  1104.5s
 => => extracting sha256:c7bef7d09442a4c4deef8e8120027e7fa3f163c7b5a0afed  4.8s
 => [internal] load build context                                          0.9s
 => => transferring context: 634B                                          0.0s
 => ERROR [2/4] RUN dnf install -y      gcc-c++     arm-none-eabi-gcc-  1935.6s
------                                                                          
 > [2/4] RUN dnf install -y      gcc-c++     arm-none-eabi-gcc-cs     arm-none-eabi-gcc-cs-c++     arm-none-eabi-binutils-cs     arm-none-eabi-newlib     cmake:
1833.1 Fedora 36 - x86_64                               47 kB/s |  83 MB     30:19                                                                              
1884.1 Fedora 36 openh264 (From Cisco) - x86_64        1.2 kB/s | 2.5 kB     00:02    
1934.7 Fedora Modular 36 - x86_64                      0.0  B/s |   0  B     00:49    
1934.7 Errors during downloading metadata for repository 'fedora-modular':
1934.7   - Status code: 404 for https://d2lzkl7pfhq30w.cloudfront.net/pub/fedora/linux/releases/36/Modular/x86_64/os/repodata/repomd.xml (IP: 18.160.94.44)
1934.7   - Status code: 404 for http://dl.fedoraproject.org/pub/fedora/linux/releases/36/Modular/x86_64/os/repodata/repomd.xml (IP: 38.145.60.24)
1934.7   - Status code: 404 for https://dl.fedoraproject.org/pub/fedora/linux/releases/36/Modular/x86_64/os/repodata/repomd.xml (IP: 38.145.60.24)
1934.7   - Status code: 404 for http://mirror.rnet.missouri.edu/fedora/linux/releases/36/Modular/x86_64/os/repodata/repomd.xml (IP: 128.206.116.77)
1934.7   - Curl error (28): Timeout was reached for http://ftp.tudelft.nl/download.fedora.redhat.com/linux/releases/36/Modular/x86_64/os/repodata/repomd.xml [Operation too slow. Less than 1000 bytes/sec transferred the last 30 seconds]
1934.8 Error: Failed to download metadata for repo 'fedora-modular': Cannot download repomd.xml: Cannot download repodata/repomd.xml: All mirrors were tried
------
Dockerfile:3
--------------------
   2 |     
   3 | >>> RUN dnf install -y  \
   4 | >>>     gcc-c++ \
   5 | >>>     arm-none-eabi-gcc-cs \
   6 | >>>     arm-none-eabi-gcc-cs-c++ \
   7 | >>>     arm-none-eabi-binutils-cs \
   8 | >>>     arm-none-eabi-newlib \
   9 | >>>     cmake
  10 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c dnf install -y      gcc-c++     arm-none-eabi-gcc-cs     arm-none-eabi-gcc-cs-c++     arm-none-eabi-binutils-cs     arm-none-eabi-newlib     cmake" did not complete successfully: exit code: 1

What do I try?

Thank you.
Randy

mikaelnousiainen, 2023-11-08:
@rpratt20 Hmm. That looks like an error downloading packages from Fedora repository. As I'm running it here locally, it downloads just fine, so maybe the mirror it had picked had issues. Fedora 36 is still being maintained until next year, so it's definitely not too old.

Another possibility is that maybe the Docker installation has some issues with the Internet connectivity, though the status code 404 doesn't suggest that.

Can you try again (after waiting 1-2 days)?

rpratt20, 2023-11-08:
I believe it worked correctly this morning. It appears to me that link resulting in the 404 error has been changed to include an /archive/ folder level. The link not functioning lead to this new release: 

Arm GNU Toolchain
Version 13.2.Rel1

Released: October 30, 2023

closed 2023-11-08 by rpratt20