mikaelnousiainen/RS41ng #74: Mod RS41ng Firmware for Serial Port Configuration

Joe-WJ2B, 2023-11-26:
Request to mod the RS41ng firmware to use the serial port on the RS41 and DFM17 to set and permanently save the operating parameters (Callsign, Mode, Rate, Freq.) instead of having to use a config file and then have to rebuild the firmware and re-flash the unit. If so, it would make the re-purposing of these units much simpler for people who are not computer programming experts since they could easily use a windows computer to flash the supplied .bin or .hex file into the re-purposed unit and then use Putty to connect to the serial port to set the config without having to know how to build the firmware. Thanks, Joe WJ2B

mikaelnousiainen, 2024-01-03:
@Joe-WJ2B Thanks for the idea. This is actually something we've been considering before and IMO it would be a really great addition. However, it requires quite a lot of effort to support all the different options in RS41ng -- and to develop some kind of "command set" for the serial port that is easy to use for the configuration. Additionally, the firmware size is quite large and already close to the maximum possible (supported by RS41), so adding this type of complex feature may get difficult in terms of the space it takes in the firmware binary.

Keeping this open if there are folks willing to contribute!

That said, I may work on this feature in the coming months, once I have a bit more time for this project -- no promises though :D

KB8RCO, 2024-08-23:
May I suggest only adding a few key parameters that can be changed through a serial interface:
  CALLSIGN
  TX Frequency for the main chip (Si4032 or Si4063) 'defaults' => there only seems to be 2. (Ex: 432.500 & 432.501)

This would probably handle most of the cases (changing CALLSIGN is probably the most common).

An option to reduce the interface considerably would be to use offsets for the different modes:
        TX_BASE_FREQ 432500000
        TX_APRS_OFFSET   0
        TX_CW_OFFSET   800
        TX_PIP_OFFSET   800
        TX_HORUS_V1_OFFSET   1000
        TX_HORUS_V2_OFFSET   1000
        TX_CATS_OFFSET   -2000000                        (or start the BASE 2 MHz lower)
Then by changing a single parameter through the serial interface, you would move all of them.  
Changing offsets would have to be done by recompiling.

Rob KB8RCO