mikaelnousiainen/RS41ng #75: DFM17 APRS packets do not format correctly

rpratt20, 2023-11-28:
The address fields and the routing fields in the packet header are transmitted in a mixed order.
Expect:
[sender callsign]>[to callsign],[WIDE1-1],[WIDE2-2],:

TRANSMITTED ORDER:
[WIDE2]>[WIDE1-1],1:

Looks as sender callsign is replaced with part of second route.  To callsign replaced with first route. Both sender and destination data are missing but the routing fields are missing from the packet.
I don't see routing declared in ax25 header file so memory pointers likely address incorrect area.

rpratt20, 2023-11-29:
I have fixed this and all works well. A couple minor fixes and a note for config.h file.

In file ax25.h the definition of ax25_packet_header can have a minor change to make char source_ssid a type uint8_t. The change aligns this definition with the definition of destination_ssid and also will then match the declaration of the uint16-t ax25_encode_packet_aprs function in file ax25.c. This does not change operation but cleans up the code.

Second change, in file ax25.c Line 92, change the handling of destination_ssid to read like the line in source_ssid.
it should read:
header->destination_ssid = (destination_ssid>= 'A' ? destination_ssid - 7 : destination_ssid);;

Finally, add a note in config.h to the APRS_RELAYS.  //There can not be a blank space in the APRS_RELAYS. Only comma field seperation.

mikaelnousiainen, 2023-11-29:
@rpratt20 Thanks for the bug report and fixes. To speed up applying the fix, could you create a pull request with the actual code changes?

Also, could you attach your full config.h file in a comment here (there is a way to attach downloadable files to these comments)?

That way I could see what causes the memory corruption.

rpratt20, 2023-11-29:
I will try a pull request tonight. I’ve never done that. I think the major issue that brought the problem to light was a space in my config path. The coding code does not handle a space between the two path entries. Not a problem if entered correctly. Otherwise, I believe the one byte ssid field was running over memory. I think my change sets destination ssid to a single byte. I had a hard time getting it working because the fields not only were wrong but they changed order. Some attempts to fix also broke transmitter operation. I ran all night last night and have received on a couple of systems today transmitting on 144.39 with a dfm17 on power set to 30. I get about 3/8 mile in town with the dfm on my dash while driving. RandySent from my iPhone

mikaelnousiainen, 2024-01-03:
@rpratt20 I believe this was fixed by your code changes in #77, closing. Thanks!

closed 2024-01-03 by mikaelnousiainen