mikaelnousiainen/RS41ng #76: Cant debug anymore - Memory limit in RS41?

whallmann, 2023-11-30:
Problem:  if i try to use the debug system, the file size increases over a limit of 64 kb.  Is there any chance to get it lower? How do you @mikaelnousiainen debug this? 

**Details**:
If i try to enable
```
#define SEMIHOSTING_ENABLE
#define LOGGING_ENABLE
```
the make command shows this infos:
```
[ 32%] Linking CXX executable RS41ng.elf
Building /home/wolfgang/RS41ng_workspace/build/RS41ng.hex 
Building /home/wolfgang/RS41ng_workspace/build/RS41ng.bin
   text	   data	    bss	    dec	    hex	filename
  65148	   1432	   5192	  71772	  1185c	RS41ng.elf
[ 78%] Built target RS41ng.elf
```
```
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.540079
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800281c msp: 0x200019e0
** Programming Started **
auto erase enabled
Info : device id = 0x10016420
Info : flash size = 64kbytes
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000003a msp: 0x200019e0
Warn : no flash bank found for address 8010000
wrote 65536 bytes from file build/src/RS41ng.elf in 2.187044s (29.263 KiB/s)
** Programming Finished **
** Verify Started **
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x200019e0
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x200019e0
Error: checksum mismatch - attempting binary compare
diff 0 address 0x08010000. Was 0xff instead of 0x00
diff 1 address 0x08010001. Was 0xff instead of 0x00
diff 2 address 0x08010002. Was 0xff instead of 0x10
diff 3 address 0x08010003. Was 0xff instead of 0x27
```
the linker shows more then 64 k so this maybe the Limit - so i cant flash to unit 

BTW: without SEMIHOST Switch the Linker shows:
```
   text	   data	    bss	    dec	    hex	filename
  60148	   1432	   5192	  66772	  104d4	RS41ng.elf
```

whallmann, 2023-12-15:
OK i selected only one mode now on 4032 Chip - not 3 at the same time to debug the mix of it. 
Then it fit into the borders of the STM32.

BAD NEWS:
Today i tried to continue on this si5351 chip to send WSPR (only this mode - no more on 4032 or so).  
While still no signal is on the output of the si5351, i enabled the 
Debug flags 

> #define SEMIHOSTING_ENABLE
> #define LOGGING_ENABLE

But unfortunately the MAKE will show the linker has problems. It will not fit anymore into memory. See here:

> [ 78%] Linking CXX executable RS41ng.elf
> /usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: RS41ng.elf section `.text' will not fit in region `rom'
> /usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 5792 bytes
> collect2: error: ld returned 1 exit status
> make[2]: *** [src/CMakeFiles/RS41ng.elf.dir/build.make:1119: src/RS41ng.elf] Fehler 1
> make[1]: *** [CMakeFiles/Makefile2:92: src/CMakeFiles/RS41ng.elf.dir/all] Fehler 2
> make: *** [Makefile:84: all] Fehler 2

I am very depressed about this, because without debuging i cant continue my project. 
Any ideas?
ps: its the latest version published on mikaelnousiainen/RS41ng

regards
Wolf DF7PN

mikaelnousiainen, 2023-12-18:
@whallmann This is a known issue with the firmware, as with the Si5351 code and all the additional modes, the code size gets very close to the RS41 limit.

Can you try to disable ALL other features and only leave the Si5351 TX enabled (with only 1 mode) and debugging enabled? That might help.

whallmann, 2023-12-18:
Hi,
nice idea but sorry, this was the first try with only one si5351 mode. This
is still enough to get the overflow message. No other modes are set true.
No chance this way to analyse why si5351 will not transmit anything.
Meanwhile i tried to use the LEDs to catch some visible infos if a code
line is reached. This way i finally cougth the info that the si5351-handler
inits the board successfully but i am not shure if 30 seconds are nessesary
for this. Within the c++ code its hard to get the led signals to address.
so i gave up for now.
cheers
Wolf

mikaelnousiainen, 2024-01-03:
@whallmann Ok, this is then indeed a more serious issue. Currently, I don't have an easy workaround for this -- other than maybe trying to remove/comment out manually some debug print statements, because all the text in them takes space.

The Si5351 library with all the digital modes supported take up a lot of space and probably should be rewritten for this purpose -- but that's a larger task...

Keeping this open if anyone has suggestions or wishes to make changes to help reducing firmware size.