mikaelnousiainen/RS41ng #84: Power OFF Function Not Working for both RS41 and DFM17

Joe-WJ2B, 2024-02-11:
When the #define ALLOW_POWER_OFF field in the config.h file is set to "true", the power OFF function is not working in both the RS41 and DFM17. I have held down the Power Button while the sonde is not transmitting but the sonde does not turn off. Only way to power OFF is to remove the batteries.

K4KDR, 2024-02-11:
Have noticed the same thing & thought perhaps that the button had to be pressed at some exact point and/or for some exact duration.   However, have not yet found the right combination so I also have had to just remove the batteries.    --Scott,  K4KDR

videobuff, 2024-04-27:
Same expierence here (27 april) Green led is flashing though, but red is off