mikaelnousiainen/RS41ng #93: CATS not transmitting from DFM-17

rpratt20, 2024-06-09:
I spent a few hours this afternoon attempting to try the new cats option. I programmed a sequence of APRS, Hv2, cats with 30 second repeating cycle. APRS and HV2 both transmitted and decoded. I could not identify a cats transmission at all. I tried the cats-sdr-software, an ht set to the frequency and an sdr# but no signal at all transmitted from the DFM. I used the GitHub cats settings other than call sign and comment text and the not balloon reporting option. The cats sdr receiving station did appear on APRS.fi along with the APRS icon from the DFM APRS transmission received through APRS.

rpratt20, 2024-06-10:
I tried a few CATS settings again this morning. With only CATS and no APRS or HV2 in the rotation the board appears to hang in boot mode. The green and yellow leds remain on steady again with no rf transmitted. Adding a combination APRS CATS rotation allows booting and APRS transmitting with the CATS never keying.

rpratt20, 2024-06-11:
This is partially solved. I found that file radio.c did not update when I changed version to include Cats. I now have transmission as scheduled but I have not achieved any decoding.

scd31, 2024-06-11:
meow meow meow!

What software/hardware are you running for decoding? Also, do you hear the CATS packets on a normal FM/SSB transceiver?

rpratt20, 2024-06-12:
Hello,I was using the cats-sdr-igate with an old rtl-sdr on Ubuntu. I’ve now re-installed drivers and connected a new rtl-sdr v3.  Later yesterday I was able to decode reliably. I plan to retry the old sdr today because it was decoding Horus v2 and regular APRS. The cats position packets appear to transmit faster than a Baofeng can produce any more than a click. Often the receive led does not blink. I suspect battery life might be greatly extended on the transmitter.  The issue I submitted can be closed as the problems appear to have been mine.RandySent from my iPhone

scd31, 2024-06-12:
Glad you were able to get it sorted!

rpratt20, 2024-06-13:
I have note to add.  With an old sdr the config.toml file must be manually set to compensate ppm error in the older sdr. There is no compensation entry.

scd31, 2024-06-14:
Yeah that sounds about right. I don't think I bothered to add a ppm option to the SDR I-Gate

Bertruss, 2026-09-18:
@rpratt20 Hey, sorry to necro this, but I can't actually get APRS working on the DFM-17 in the latest version of the firmware... the packets transmit and decode but don't actually have valid GPS coordinates attached. Did you have that working?

rpratt20, 2026-09-18:
I’ve done very little with the latest version. I thought the one unit I programmed initially worked OK. I have several personal changes in the older RS41ng so I am continuing to use the Horus V2 previous software for what I am doing. I did note that time synch and transmit offset time function quite differently in the new version. Also, if you use any gps setting other than 1000 milliseconds and 1sec for reporting updates you will likely not function correctly with either version of the software. You might get lucky on a specific setting but it will take many trials to get working timing combinations. I just reread your question and I’m not sure if you refer to CATS subject or APRS text? One other note: if you have a space in the APRS PATH it won’t function. “WIDE-1,WIDE-2” or just “” for example.  Hope some of this helped. RandySent from my iPhone