mikaelnousiainen/RS41ng #98: The time does not update when I use solar panels.

GianlucaB2000, 2024-09-29:
Hi,
I have been testing the radiosonde using six solar panels for the past few days. A few hours after sunrise, when there is enough light for the radiosonde to work, I encounter a problem where it keeps sending the same time repeatedly. This prevents the position from being sent to Sondehub.

However, since the radiosonde is still able to send GPS position data without any problems, the firmware does not restart the radiosonde. I think that restarting the radiosonde for example every couple of hours, might solve the problem, since the capacitor is already charged enough for the GPS to work properly.

Here is an example of the error from Horus GUI: 2024-09-29 10:22:09,721 ERROR: Sondehub Amateur Uploader - Payload and receiver times are more than 3 minutes off. Either the payload does not have GNSS lock or the system time is not set correctly. Uploading not in progress.

rpratt20, 2024-12-21:
Is this for DFM or RS41?

GianlucaB2000, 2024-12-21:
I used an Rs41, I tried various settings to reduce the consumption of the
radiosonde, but I was unable to solve the problem

rpratt20, 2024-12-22:
You might check the Whallmann fork of the project.Sent from my iPhone