mikaelnousiainen/RS41ng #103: NEW PCB RS41

iz2fly, 2025-01-08:
Warning! There are new PCBs for RS41 with a new 64 pin processor instead of 48 pin. STM32L412.
This firmware will need to be modified.
[image]

Nevvman18, 2025-01-26:
Also this alternative firmware can be utilized:
https://github.com/Nevvman18/rs41-nfw

mikaelnousiainen, 2025-02-19:
We're working on supporting the new RS41 PCB / revision, but adding support for the new microcontroller turned out to be more difficult since the hardware abstraction layer (C library) RS41ng uses cannot be used with this new microcontroller - so some wrapper layers need to be added. Anyway, stay tuned!